ucb-substrate/sram22 #360: README example design fails

mguthaus, 2022-12-08:
I'm running the main branch without the BWRC tools. The creation fails at netlist conversion:

~/sram22/sramgen/sramgen_sram_32x32m2w8_replica_v1 (master)$ sramgen -o .

 ________  ________  ________  _____ ______     _______   _______
|\   ____\|\   __  \|\   __  \|\   _ \  _   \  /  ___  \ /  ___  \
\ \  \___|\ \  \|\  \ \  \|\  \ \  \\\__\ \  \/__/|_/  //__/|_/  /|
 \ \_____  \ \   _  _\ \   __  \ \  \\|__| \  \__|//  / /__|//  / /
  \|____|\  \ \  \\  \\ \  \ \  \ \  \    \ \  \  /  /_/__  /  /_/__
    ____\_\  \ \__\\ _\\ \__\ \__\ \__\    \ \__\|\________\\________\
   |\_________\|__|\|__|\|__|\|__|\|__|     \|__| \|_______|\|_______|
   \|_________|

Starting SRAM generation...

Configuration file: "/home/mrg/sram22/sramgen/sramgen_sram_32x32m2w8_replica_v1/sramgen.toml"
Output directory: "/home/mrg/sram22/sramgen/sramgen_sram_32x32m2w8_replica_v1"

SRAM parameters:
        Number of words: 32
        Data width: 32
        Mux ratio: 2
        Write size: 32
        Control mode: ReplicaV1

Tasks:
  Done             [1/4] Generate plan
⠁ In Progress      [2/4] Generate netlist
⠁ Pending          [3/4] Generate layout
⠁ Pending          [4/4] Generate Verilog                                                                                                                                                               Traceback (most recent call last):
  Done             [1/4] Generate plan
  Failed           [2/4] Generate netlist
  Skipped          [3/4] Generate layout
  Skipped          [4/4] Generate Verilog

Error: Error converting netlists to SPICE format

Caused by:
    Netlist generation script failed with status Some(1)

rahulk29, 2022-12-09:
Updated the README with some additional things you need to do to set up dependencies. Let me know if it works now!

mguthaus, 2022-12-09:
I installed the python dependencies, but I still get the same error.

rahulk29, 2022-12-10:
There's one more Python dependency that needs to be installed. Updated the README with instructions on how to install it.

rahulk29, 2022-12-16:
@mguthaus Is this still an issue?

mguthaus, 2022-12-16:
This fixed the issue. Thanks!

rahulk29, 2022-12-16:
Great!

closed 2022-12-16 by rahulk29