ucb-substrate/sram22 #414: psf-ascii changed to psfparser

zzdywc, 2023-05-12:
due to the change of psf-ascii, now sram22 isn't word anymore. I tried fork psfpaser and reset to 3 weeks ago, but still not work, so what is the workble psf-ascii commit?

jackpanderson, 2023-05-12:
Seconded, tried changing psf-ascii to psfparser in the Cargo.toml and ran into more issues where it expects psf-ascii.

zzdywc, 2023-05-13:
Finally ，I found the right commit，it was “33951c7041024e7cc230208d4af3062ad6bcc749”。I tried some parameter of sram, but it's not very powerful, for example word size = 64, num word = 58, will fail. thesram22  is very fast. i think it's due to no drc process.

rahulk29, 2023-05-16:
Thanks for trying out SRAM22! The psfparser issue should be fixed by #415.

We currently only support power-of-2 SRAM sizes, but we may eventually support non-power-of-2 sizes. SRAM22 doesn't run DRC by default, but users with access to the commercial tool plugins can tell SRAM22 to run DRC via Calibre. Unfortunately, the open source tools don't seem to understand the special SRAM DRC rules, so there is no option to run DRC with magic/klayout/etc.