ucb-substrate/sram22 #444: Fail to generate layout. "assertion failed: nx >= 0 && ny >= 0"

lu0de, 2023-10-07:
(base) [luode@localhost sram22_64x32m4w8]$ sram22 -o .

 ________  ________  ________  _____ ______     _______   _______
|\   ____\|\   __  \|\   __  \|\   _ \  _   \  /  ___  \ /  ___  \
\ \  \___|\ \  \|\  \ \  \|\  \ \  \\\__\ \  \/__/|_/  //__/|_/  /|
 \ \_____  \ \   _  _\ \   __  \ \  \\|__| \  \__|//  / /__|//  / /
  \|____|\  \ \  \\  \\ \  \ \  \ \  \    \ \  \  /  /_/__  /  /_/__
    ____\_\  \ \__\\ _\\ \__\ \__\ \__\    \ \__\|\________\\________\
   |\_________\|__|\|__|\|__|\|__|\|__|     \|__| \|_______|\|_______|
   \|_________|

Reading configuration file...

Configuration file: "/home/luode/temp/sram22_64x32m4w8/sram22.toml"
SRAM parameters:
        Number of words: 256
        Data width: 32
        Mux ratio: 4
        Write size: 8
        Control mode: ReplicaV2

Tasks:
  Done             [1/4] Generate plan
  Done             [2/4] Generate netlist
⠁ In Progress      [3/4] Generate layout
⠁ Pending          [4/4] Generate Verilog                                                                                                                                               Warning: Importing empty port: vpb
Warning: Importing empty port: vnb
Warning: Importing empty port: vpb
Warning: Importing empty port: vpb
thread 'main' panicked at /home/luode/.cargo/git/checkouts/substrate-86fbb4e3c8030876/82a793a/substrate/src/layout/routing/auto/mod.rs:108:9:
assertion failed: nx >= 0 && ny >= 0
stack backtrace:
   0: rust_begin_unwind
             at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/std/src/panicking.rs:595:5
   1: core::panicking::panic_fmt
             at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/core/src/panicking.rs:67:14
   2: core::panicking::panic
             at /rustc/cc66ad468955717ab92600c770da8c1601a4ff33/library/core/src/panicking.rs:117:5
   3: substrate::layout::routing::auto::GreedyRouter::with_config
   4: sram22::blocks::control::layout::<impl sram22::blocks::control::ControlLogicReplicaV2>::route
   5: <sram22::blocks::control::ControlLogicReplicaV2 as substrate::component::Component>::layout
   6: substrate::data::SubstrateCtx::instantiate_layout
   7: sram22::blocks::sram::layout::<impl sram22::blocks::sram::SramInner>::layout
   8: substrate::data::SubstrateCtx::instantiate_layout
   9: <sram22::blocks::sram::Sram as substrate::component::Component>::layout
  10: substrate::data::SubstrateCtx::write_layout
  11: sram22::plan::execute_plan
  12: sram22::cli::run
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

barakhoffer, 2024-01-09:
Same here. any solution for this?

rohanku, 2024-01-11:
Sorry for the late reply. We aren't able to reproduce this issue locally, it seems like generating a 256x32m4w8 SRAM works fine on the latest version. Could you try running `cargo update` and `cargo clean`, then rerunning the installation steps? It is possible that you are using an old version of Substrate since git dependencies are not semantically versioned.

rohanku, 2024-01-11:
Actually, after a little bit more thought it likely is an issue of using the standard SKY 130 PDK. We made a couple modifications to the PDK to allow Substrate to read the standard cells correctly, so you will need to clone our version of the PDK instead: https://github.com/ucb-substrate/skywater-pdk.

barakhoffer, 2024-01-11:
Yes, it fixes the problem. thanks!

closed 2024-01-11 by rohanku