ucb-substrate/sram22 #451: Project questions w.r.t SKY 130 tapeout

c-93, 2024-04-17:
- As we consider SRAM22 for an asic design: Was SRAM22 already taped-out in SKY130 before?

- Would it be possible for external users, who have access to corresponding licenses (or open source tools) to sram22 macros? One could try magic, netgen, and klayout.

- Offtopic: How do the PDK patches conflict with the upstream SKY130 PDK? Would it be possible / a good idea to use the patched PDK just for a SRAM22 macro, and the upstream PDK for the remaining design? Or should the patched PDK be used globally?

Thanks for this great project. :)

rahulk29, 2024-04-26:
Thanks for checking out SRAM22!

- I'm afraid SRAM22 isn't quite ready for general use -- we are working to fix a few bugs found on test chips, and will publish updates when they are ready (likely a few months).
- I am not sure the open source tools understand SRAM DRC rules, so DRC/LVS would likely be challenging with open source tools. SRAM22 macros are DRC/LVS clean in Calibre, if you have access to that.
- The patched PDK have a few minor modifications, such as connecting VPB to VPWR in tap cells. I would recommend using the regular PDK for your design, and our patched PDK just for SRAM22.

c-93, 2024-04-29:
Thanks for your response. I wish you all the best. You may close the issue or leave it open for documentation.

RTimothyEdwards, 2025-04-13:
@rahulk29 :  The `sram22_sky130_macros` repository latest commits are from only a few months ago.  Does that mean that for those layouts, the issues mentioned above (one year ago) have been resolved?

rahulk29, 2025-04-13:
The most recent macros have fixes that we believe address issues with older versions, but we do not yet have silicon measurement results to demonstrate that they work. We had a test chip on one of the eFabless ChipIgnite runs, though it's not clear if/when we might get that chip back. We're tentatively planning to resubmit the test chip on Cadence's MPW service.

In short, I wouldn't yet consider the latest macros production ready.

RTimothyEdwards, 2026-11-13:
@rohanku , @rahulk29 :  I want to be building a new harness chip design for my Chipalooza program and I want to use a 1024x8 SRAM.  I would very much like to add your SRAM blocks to the sky130 open PDK required libraries for open-pdks (and move the unmaintained openRAM macros to "optional").

ranaumarnadeem, 2026-11-15:
Hi, I maintain autoMBIST (https://github.com/ranaumarnadeem/autoMBIST), an open-source memory BIST and repair generator that currently targets OpenRAM sky130 macros. I'd like to help with the open-tool verification question here, following on from #505: a Magic/Netgen DRC/LVS stage for SRAM22 macros that runs without Calibre.

One thing I've seen that I expect applies here too: the stock sky130 Magic setup reports about a million DRC errors on OpenRAM macros, and LVS fails on split nets, even for macros that pass OpenRAM's own Magic/Netgen check. OpenRAM only gets a clean result with its own SRAM-specific handling of the bitcells, so SRAM22 would need the same for its bitcell.

@RTimothyEdwards (I hope you remember me from ORconf), if SRAM22 is going into open_pdks, is open-tool DRC/LVS of the macros already part of that plan? I'd rather help with that than start something in parallel. If not, would a PR adding a Magic/Netgen verification stage next to the Calibre one be welcome?

RTimothyEdwards, 2026-11-15:
@ranaumarnadeem :  My original intent was to not try to capture the SRAM DRC rules in magic partly because they are not defined anywhere, although I could simply make an exception for the rules that fail within the SRAM block.  Since I wrote the magic techfile for sky130, I have added `exception` and `exemption` rules in magic which I could apply here.  That would provide a clean DRC for the SRAM core cells.

ranaumarnadeem, 2026-11-16:
@RTimothyEdwards Thanks, an exemption for the SRAM core cells sounds like the right fix. To help, I can run the current sky130 Magic DRC over a few macros from sram22_sky130_macros and post which rules fail and where, so it's clear which exceptions are needed and we can check the result afterwards.

Once DRCs are done I believe then we can move onto LVS