ucb-substrate/sram22 #479: issue while trying to install

goatgate, 2024-10-22:
| #[cfg(feature = "commercial")]
| ^^^^^^^^^^^^^^^^^^^^^^
|
= note: expected values for feature are: default
= help: consider adding commercial as a feature in Cargo.toml
= note: see <https://doc.rust-lang.org/nightly/rustc/check-cfg/cargo-specifics.html> for more information about checking conditional configuration
warning: unexpected cfg condition value: commercial
--> src/lib.rs:99:15

rohanku, 2024-11-30:
This is just a warning, so it should be fine if you see that while installing. Did you see any other issues while trying to install? We recently pushed some substantial changes, so it is possible that you will no longer see these issues with the latest version.

goatgate, 2024-12-02:
tried to reinstall, here's the error i'm getting:error: environment variable `SKY130_OPEN_PDK_ROOT` not defined at compile time
  --> src/lib.rs:45:40
   |
45 | pub const SKY130_OPEN_PDK_ROOT: &str = env!("SKY130_OPEN_PDK_ROOT");
   |                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: use `std::env::var("SKY130_OPEN_PDK_ROOT")` to read the variable at run time
   = note: this error originates in the macro `env` (in Nightly builds, run with -Z macro-backtrace for more info), 
   
   
   Just to clarify, do i need to individually install openpdks for sram22?

rohanku, 2024-12-03:
I think you might be missing the last dependency from the README:

> In order to use SRAM22, your system will need to have the following components:
> 
> - Rust (SRAM22 is tested with version 1.81.0)
> - Make
> - A local clone of our [slightly modified version of the SKY 130 PDK](https://github.com/ucb-substrate/skywater-pdk). You will also need to set the environment variable SKY130_OPEN_PDK_ROOT to the absolute path of the local PDK's root directory. Substrate uses standard cells from the sky130_fd_sc_hs library, so you will also need to run the following from the PDK root directory:
> 
>   ```
>   git submodule update --init libraries/sky130_fd_sc_hs/latest
>   ```

closed 2025-05-23 by rahulk29