ucb-substrate/sram22 #504: Missing boundary

urish, 2025-08-14:
First of all, thanks for this super cool project :-)

I'm trying to integrate the [sram22_512x8m8w1](https://github.com/rahulk29/sram22_sky130_macros/tree/master/sram22_512x8m8w1) macro into an experimental Tiny Tapeout project. It almost works - but magic chokes when it tries to read the GDS file, complaining about missing boundary:

```
Failed to extract PR boundary from GDSII view of macro 'sram22_512x8m8w1'. Ensure that the GDSII view has a PR boundary layer.
```

I looked into the GDS and it seems like the prBoundary.boundary doesn't contain any shape, only texts. Is that on purpose?

urish, 2025-08-20:
@RTimothyEdwards can you please explain the role of prBoudnary and why magic needs it when reading the GDS?

RTimothyEdwards, 2025-08-20:
@urish :  It is a place & route boundary marker, so it defines the placement bounding box of a cell.  There is nothing in GDS that will define a cell boundary other than a layer, so it's necessary to define a GDS layer number to represent a cell boundary.

That said, it is not supposed to be _mandatory_ to have a cell boundary (the extent of geometry will be used for the boundary if there is no other boundary definition).  My suspicion is that somehow the presence of the boundary layer for text is resulting in magic thinking that there are zero-area boundaries being defined.  That is likely just an oversight in the way the boundary layer is being handled.

(Thanks for pinging me, as I didn't see this issue when it was created, since I am not following this repository.)

RTimothyEdwards, 2025-08-20:
Based on Sylvain's comment above:

There is a 2nd method which is probably the better one to use, which is that if a LEF file exists of the cell, then read the LEF view on top of (i.e., after) the GDS view.  That will annotate the GDS view with, among other things, the cell size as defined in the LEF, which it will do by creating the FIXED_BBOX property.

Only if a LEF file does not exist and there is no FIXED_BBOX property, should the extent of geometry be used as the default bounding box.

RTimothyEdwards, 2025-08-20:
@urish :  Magic version 8.3.544 (just pushed to opencircuitdesign.com) fixes this problem.  Please test on the updated version of magic, as that may resolve the issue with the LibreLane script.

smunaut, 2025-08-20:
It won't help here because LibreLane only does `gds read` it doesn't annotate with the LEF but I guess it could be made to.